Rightpoint/RZBluetooth #5: Manually prompt discovery of characteristics?

drdaz, 2016-01-14:
Hey there

Is there some way of manually making characteristic discovery happen? I have a write characteristic that I'd like to have discovered before I write to it...

And a hacky solution is fine, because I'm writing a hack.

KingOfBrian, 2016-01-14:
Hey @drdaz -- I made a branch with an implementation of manual characteristic discovery. Can you try it out? I haven't been able to add a test for it, but it should "Just Work" :tm:

Thanks!

drdaz, 2016-01-15:
Oh wow thanks :)

I'll try merging that into my fork asap.

drdaz, 2016-01-29:
Hey! Just an FYI. I've been using this patch for a few weeks now. Seems to Just Work.

KingOfBrian, 2016-02-08:
I merged in the fix, so I'm closing this out.

closed 2016-02-08 by KingOfBrian