Rightpoint/RZBluetooth #8: Crash when testing...

drdaz, 2016-01-15:
I'm using the library in an XCTest, and consistently crash here. It's pretty bizarre; centralManager seems to be a RZBCentralManager (see debug console), but the assert fails anyway...

[image]

KingOfBrian, 2016-01-15:
That makes no sense! The only thing I can think of is that the centralManager has been released so the runtime thinks the object is dead, but the debugger is still able to talk to it. Can you double check that  the central manager is strongly held by the test class?

If it's not something like that, it looks like a bug in NSAssert, which I obviously don't think is the case.

KingOfBrian, 2016-01-15:
If that's not the case, try copying the unit test case into RZBluetooth and try to get it to reproduce there.

KingOfBrian, 2016-01-15:
Also, make sure that you're not running in Release mode, and make sure there's no optimizations turned on. Default Debug settings should be fine, but optimizations can make the debugger tell terrible lies.

drdaz, 2016-01-18:
So this is pretty confusing. I can't reproduce this, except for all the time in the test in my product. If I make a separate tester project, and construct the same situation there's no problem.

The central manager is strongly held by its containing class. As a property no less.

To make things even more confusing, I've added the same assert into my code immediately before the call to rzb_discoverCharacteristicUUIDs (which is where we fail), and it passes fine. I'm really at a loss here.

drdaz, 2016-01-18:
Finally I managed to reproduce the issue! Run testExample in RZBluetoothTest0rTests.m, and watch in amazement as true != true.
[RZBluetoothTest0r.zip](https://github.com/Raizlabs/RZBluetooth/files/94778/RZBluetoothTest0r.zip)

drdaz, 2016-01-19:
So this is probably going to go down as an unsolved mystery.

I've worked around this by separating my BT service out into a framework, and creating a thin app to run my tests through. This makes good sense now too, since the BT comms layer now needs to be used in a second app.

closed 2016-01-19 by drdaz

KingOfBrian, 2016-01-19:
Pretty strange, I couldn't get anything to crash. The test didn't have any simulation environment so it would only have a chance to pass if ran on device with the peripheral near by.

 If you're writing a test, there's a subclass provided by the `Test` sub spec which should help get the simulated bluetooth environment going.

Glad the issue isn't blocking you anymore though!

drdaz, 2016-02-03:
> If you're writing a test, there's an XCTestCase subclass provided by the Test sub spec which should help get the simulated bluetooth environment going.

I just started thinking about test again, and this seems important. Can you offer any more detail?

KingOfBrian, 2016-02-03:
You should be able to copy the approach used by the battery level tests.  `RZBProfileBatteryTests` (and the base class) perform all of the testing, and `RZBSimulatedDevice+RZBBatteryLevel` implements the simulated battery level API's.  Then your test target would need to include the `RZBluetooth/Test` pod.

Sorry for the testing section in the readme.... it is terrible :disappointed:

drdaz, 2016-02-04:
So do I need to use the profile level APIs to use this test strategy?

I've used RZBluetooth in a service, so it's buried behind my code. I'd like to use the simulated device in place of the physical device, and test the whole service. Is this possible? Would I need to use some kind of dependency injection to hit the simulated device?

drdaz, 2016-02-08:
Should the startAdvertising method on RZBSimulatedDevice work? I'm getting a crash since self.advertisedServices is nil at call time.

This feels like the wrong place to be discussing this; is there a more appropriate place we could converse about this? I'm very interested in hearing how this is supposed to work. I'd be willing to try and contribute some documentation for the test functionality if I could really understand how to use it :)