Rightpoint/RZBluetooth #9: cancelConnection fires completion handler before disconnection

drdaz, 2016-01-15:
I suspect this isn't an issue with your library per se; I'm seeing the same behaviour with similar libraries. But I've been banging my head against this for a little while and I'd like to share the problem.

When the completion handler for cancelConnectionFromPeripheralUUID fires, I expect (wrongly) the peripheral to have actually disconnected. If I attempt to reconnect at this point, things get generally messy.

Is there any way of being notified when we actually have finished disconnecting?

drdaz, 2016-01-15:
Actually I seem to be off base on this; the value of state when the completion handler fires actually is disconnected. My physical peripheral shows that it's in a connected state for a few more seconds usually. I'll have to dig at this some more...

KingOfBrian, 2016-01-18:
Disconnection maintenance is super important, and varies widely from device to device. Keep looking into this, and what type of behavior you'd like, and if there's anything that I can do to help things out. I was contemplating adding a backoff reconnection, but my device doesn't need it so I couldn't test it too well.

I'm going to close this for now, as it doesn't seem to be an issue.

closed 2016-01-18 by KingOfBrian