Rightpoint/RZBluetooth #28: Replacement for Dynamic methods in RZBMockPeripheral?

drdaz, 2016-05-10:
Hi there

I just updated my version of RZBluetooth from 0.1 to 1.0.2. There were some pretty big changes, but the migration generally wasn't tough.

One thing I have lost now, that I'm not sure how to replicate, is that I was using the methods in the "Dynamic" interface extension to RZBMockPeripheral in my tests. In particular, I had implemented rub_writeData... for my test to post a notification containing the packet being written, so that I could catch the packets sent in a flow and check they were correct at a later stage.

Is there any way I can achieve something similar now?

drdaz, 2016-05-10:
... I've tried adding a write callback using addWriteCallbackForCharacteristicUUID:... but it seems that the block I provide is never called. All my calls to writeData are without responses...

drdaz, 2016-05-11:
Should I be using a mock framework with your mocks? Really quite lost here..

drdaz, 2016-05-11:
To ensure I've been clear, I'm running my test from an RZBSimulatedTestCase. I'm firing some calls through my service layer, and I would like to catch and examine the packets that arrive at the RZBPeripheral / RZBMockPeripheral.

In previous versions I just implemented rzb_writeData... to deal with the incoming packets appropriately. This approach no longer works.

It's hard to tell exactly what's going on here; depending on where in the test I stop and debug, there is either an RZBPeripheral or an RZBMockPeripheral to look at.

drdaz, 2016-05-11:
And you can ignore all this.

In some of my tests I wasn't doing things quite as the updated docs say. The callbacks I add work just fine now.

So yeah... remember to RTFM. Twice.

closed 2016-05-11 by drdaz

KingOfBrian, 2016-05-11:
Hey @drdaz , sorry to not get back to you yesterday -- this fell off my plate. Glad that the docs worked for you, they went through a bit of an overhaul. If any areas could use some enhancement, please file another issue!

Brian