Rightpoint/RZBluetooth #29: Simulating a scan and a connection

drdaz, 2016-05-12:
Hi there

One thing that isn't obvious to me from the documentation is how to simulate a scan followed by a connection.

I'm using a RZBSimulatedTestCase subclass. I'm calling my service layer, which initiates a scan. I have added services to my simulated device and called startAdvertising before the scan starts. But the scan never finds a device.

What steps am I missing?

KingOfBrian, 2016-05-13:
Hey @drdaz -- I'm not sure. It sounds like you're doing the right thing -- similar to `testScanForDevices`. From looking at the test, we call startAdvertising after the scan starts, but I just moved it and it works on either side. The best way to debug issues is to use `RZBSetLogHandler` and print out everything that comes it's way. If you paste the log from that here I can probably give you a bit more information.

drdaz, 2016-05-20:
Alright... I'm actually getting very little from the logs:

2016-05-20 14:11:43.526 xctest[4846:21269014] RZBLog: PeripheralManager is not discoverable
2016-05-20 14:11:46.559 xctest[4846:21268800] RZBLog: Ignoring RZBMockPeripheralManager updateValue since the peripheral is not connected
2016-05-20 14:11:47.560 xctest[4846:21268800] RZBLog: Ignoring RZBMockPeripheralManager updateValue since the peripheral is not connected

The "PeripheralManager is not discoverable" message comes, oddly enough, from this method: - (void)mockPeripheralManager:(RZBMockPeripheralManager *)peripheralManager startAdvertising:(NSDictionary *)advertisementData

... FWIW I'm getting these logs by doing this at the start of my test:
    RZBSetLogHandler(^(RZBLogLevel logLevel, NSString *format, va_list args) {
        NSLog(@"RZBLog: %@", format);
    });

I presume that dumps everything.

Am I supposed to call the scanCallback dispatch method at some point in the flow for things to work?

KingOfBrian, 2016-05-20:
My mistake on `PeripheralManager is not discoverable` -- I'll change that to `is discoverable`. I got the logic in my logs backwards... Also, the log handler you added isn't displaying all of the values, this should work:

```
    RZBSetLogHandler(^(RZBLogLevel logLevel, NSString *format, va_list args) {
        NSLog(@"RZBLog: %@", [[NSString alloc] initWithFormat:format arguments:args]);
    });
```

I would expect to see more in the log as well. A few things that I was looking for was:

- The peripheralManager should change state to `.PoweredOn`
- The centralManager should change state to `.PoweredOn` 
- The peripheral should make a connection or scan attempt.

The first two things are performed by `RZBSimulatedTestCase.setUp`. Are you sure you're calling super in the `setUp` method of you test case? Actually, what may be happening is your enabling the logging after that's ran. If you can move that `RZBSetLogHandler` to your `setUp` method before you call `super.setUp` we should see more information.

Assuming that `RZBSimulatedTestCase.setUp` is ran and working, the other part that I'm not seeing is a connection or scan attempt by your service layer. Are you waiting for the callback to fire in your tests? It could be that your making the service call and then the test ends and tears things down before the connection attempt is actually made.

You shouldn't need to use scanCallback until you start testing error injection.
If you want to post a gist with your test or email them to me (brian.king@raizlabs.com) I can see if I see anything else.

Brian

drdaz, 2016-05-20:
You're right; I was setting the log handler in my test method. This is what I get now:

```
2016-05-20 15:23:39.194 xctest[5508:21473148] RZBLog: Background central support is not enabled. Add 'bluetooth-central' to UIBackgroundModes to enable background support
2016-05-20 15:23:39.194 xctest[5508:21473148] RZBLog: centralManagerDidUpdateState: - <RZBMockCentralManager: 0x791426e0>
2016-05-20 15:23:39.194 xctest[5508:21473148] RZBLog: State=5
2016-05-20 15:23:39.195 xctest[5508:21473148] RZBLog: peripheralManagerDidUpdateState: - <RZBMockPeripheralManager: 0x79041d20>
2016-05-20 15:23:39.195 xctest[5508:21473148] RZBLog: State=5
2016-05-20 15:23:39.304 xctest[5508:21473254] RZBLog: PeripheralManager is not discoverable
2016-05-20 15:23:44.388 xctest[5508:21473148] RZBLog: Ignoring RZBMockPeripheralManager updateValue since the peripheral is not connected
2016-05-20 15:23:45.420 xctest[5508:21473148] RZBLog: Ignoring RZBMockPeripheralManager updateValue since the peripheral is not connected
```
That first line looks like it might be important...

I do wait for the connection to complete.

I'll see if I can send you some more details about my test...

KingOfBrian, 2016-05-20:
RZBluetooth isn't getting the connection attempt. You should see something like this:

```
2016-05-20 09:05:09.185 xctest[45929:6314055] RZBLog: connectPeripheral:C7941F53-1D3D-4E10-9095-34380E69CC3A options:(null)
2016-05-20 09:05:09.186 xctest[45929:6314055] RZBLog: centralManager:didConnectPeripheral: - <RZBMockCentralManager: 0x7c95b970> C7941F53-1D3D-4E10-9095-34380E69CC3A
```

I'd step through your code and make sure you're actually getting to the RZBluetooth connect request.

drdaz, 2016-05-20:
It's definitely not getting to the connection attempt. I'm scanning for peripherals (centralManager scanForPeripheralsWithServices:...), not finding any (timeout after 5 secs), and aborting.

KingOfBrian, 2016-05-20:
The scan should show something like this:

```
2016-05-20 10:16:41.571 xctest[46611:6344154] RZBLog: scanForPeripheralsWithServices:@[(null)] options:(null)
2016-05-20 10:16:41.571 xctest[46611:6344154] RZBLog: centralManager:didDiscoverPeripheral:advertisementData:RSSI: - <RZBMockCentralManager: 0x79018950> 8ACE536F-D5FB-46AF-96F2-6D1B7492422F
2016-05-20 10:16:41.571 xctest[46611:6344154] RZBLog: advertisementData=(null)
2016-05-20 10:16:41.572 xctest[46611:6344154] RZBLog: RSSI=(null)
```

The 2 next steps I'd take are:

1) Step into the scan call and make sure it's getting invoked
2) Print out the test's centralManager and your Service layer's centralManager and make sure they are the same object and that they're the mocked object. I think that this may be what's going on. If they're different, override `RZBSimulatedTestCase.configureCentralManager` and instead of creating a new central manager, assign `.centralManager = servicelayer.centralManager`.

If neither of these dig anything up, move the `RZBSetLogHandler` method to a `+ (void)load` -- this will make sure that anything that's happening to show up.

drdaz, 2016-05-20:
You're on to something with the centralManager I think. They're both the same object, but it's an RZBCentralManager. Not an RZBMockCentralManager...

I've called RZBEnableMock(YES) in my setUp method. I've tried both before and after [super setUp]... still getting what looks like a normal object.

KingOfBrian, 2016-05-20:
Bah, naming is complicated, and I missed some nuance here -- your RZBCentralManager has a `centralManager` property. The `centralManager` property should be a mock, and the same mock as your test is configured with.

drdaz, 2016-05-20:
I think we've found the issue, although both RZBCentralManagers' centralManager properties are nil. Both coreCentralManager properties are mocks though. Different ones.

Unfortunately fixing it will have to wait until Monday.

Thanks so much for the help, and have a great weekend! :)

drdaz, 2016-05-20:
Okay so I couldn't wait to try a fix...

My service is only instantiated during the test, so I tried setting RZBSimulatedTestCase's centralManager to the one in my service after the fact (rather than overriding configureCentralManager). Unfortunately the result is the same; and nothing new in the log.

KingOfBrian, 2016-05-20:
Over ride configureCentralManager -- `RZBSimulatedTestCase.setUp` depends
on the centralManager being configured in that method. If you look at the
implmentation you can see why -- it's not doing much, but it needs to be
powered on.

drdaz, 2016-05-23:
Superb. That did it... thanks @KingOfBrian!

closed 2016-05-23 by drdaz