Rightpoint/RZBluetooth #34: Convert to CBManagerState for iOS 10.

cpatterson, 2016-09-12:
In the iOS 10 SDK, they've combined CBCentralManagerState and CBPeripheralManagerState into one type called CBManagerState.

Currently, building RZBluetooth in Xcode 8/iOS 10 SDK throws warnings about the type conversion:

`warning: implicit conversion from enumeration type 'CBManagerState' (aka 'enum CBManagerState') to different enumeration type 'CBCentralManagerState' (aka 'enum CBCentralManagerState') [-Wenum-conversion] `

cpatterson-lilly, 2016-11-30:
Just updated to RZBluetooth version 1.1.1; I'm still seeing some build warnings related to CBManagerState caused by RZBErrors.h and RZBMockPeripheralManager.h.

KingOfBrian, 2016-11-30:
Looks like I missed the `state` property in `RZBMockPeripheralManager.h`. 

I'm really confused why I'm not getting any deprecation warnings when I build the project. Do you not get any errors in `RZBCentralManager.m` @cpatterson ?

cpatterson-lilly, 2016-11-30:
No errors, only build warnings. 

Note that I'm building a framework using CocoaPods, that adds RZBluetooth as a dependency in its own Podfile. It also uses Mock and Test. Here's my framework's Podfile:
```
# Defines a global platform for your project
platform :ios, '9.0'

# Uncomment this line if you're using Swift
use_frameworks!

target 'MyFramework' do
    pod 'RZBluetooth',      '~> 1.1'
    pod 'RZBluetooth/Mock', '~> 1.1'
end

target 'MyFrameworkTests' do
    pod 'RZBluetooth',      '~> 1.1'
    pod 'RZBluetooth/Mock', '~> 1.1'
    pod 'RZBluetooth/Test', '~> 1.1'
end
```

I've not yet attempted to build my app that includes my framework yet. If errors pop up at that point I'll let you know.

KingOfBrian, 2016-11-30:
Can you point to `bugfix/iOS10DefineChanges` real quick and double check that it's error free? I should be able to push up a fix pretty quick

cpatterson-lilly, 2016-11-30:
That did the trick.

PR #37 (2016-10-11): iOS 10 Support
Fixes 2 issues with iOS 10
- Building with Xcode 8 had a few warnings wrt `CBCentralManagerState` -> `CBManagerState`
- Unit Tests incorrectly used CBUUID

This should resolve #34
closed 2016-12-22 by KingOfBrian