Rightpoint/RZBluetooth #36: Unable to connect to peripheral if maintainConnection = TRUE and Bluetooth is power cycled

MrGoPoint, 2016-10-04:
- scanForPeripheralsWithServices:
- readCharacteristicUUID
- enableNotifyForCharacteristicUUID:
- characteristic is updated
- power off peripheral
- after RZBPeripheralStateEventDisconnected, set self.peripheral.maintainConnection = TRUE;
- Bluetooth Off/On
- After CBManagerStatePoweredOn:, scanForPeripheralsWithServices:
- peripheral is found but readCharacteristicUUID always times out with no connection

KingOfBrian, 2016-10-10:
Hey @MrGoPoint ,
   Thanks for the issue. Can you attach log messages? If you configure a log handler with `RZBSetLogHandler` and print to NSLog, it will log all CoreBluetooth delegate actions and a few RZBluetooth specific things and will help figure out what's going on. Sorry I should of mentioned this the first time.

MrGoPoint, 2016-10-12:
Will do. Any idea how to set the log handler using DDLog (Lumberjack)? Objective-c notation.

 

Thanks!

 

 

Best regards,

Brennan Hamilton

GoPoint Technology
2609 Crooks Rd
Troy, MI 48084

cell:    (248) 890-1007

Website:  GoPointTechnology.com

LinkedIn:  Brennan Hamilton

VCard:      Brennan Hamilton.vcf

KingOfBrian, 2016-10-12:
I think something like this in your app delegate will do it (Don't actually have CocoaLumberjack on any projects atm)
```
    RZBSetLogHandler(^(RZBLogLevel logLevel, NSString *format, va_list args) {
        if (logLevel != RZBLogLevelWriteCommandData && logLevel != RZBLogLevelDelegateValue) {
            DDLog(@"%@",  [[NSString alloc] initWithFormat:format arguments:args]);
        }
    });
```

This won't include any of the data transmitted, but will include all of the UUID's you use. I'm assuming that's not a big concern, but something to be noted. If you'd feel more comfortable emailing it to me, that's fine too.

Thanks!

MrGoPoint, 2016-10-12:
Works. I’ll roll back the app and get a log file for you.

 

 

 

Best regards,

Brennan Hamilton

GoPoint Technology
2609 Crooks Rd
Troy, MI 48084

cell:    (248) 890-1007

Website:  GoPointTechnology.com

LinkedIn:  Brennan Hamilton

VCard:      Brennan Hamilton.vcf

KingOfBrian, 2016-10-17:
Any update on this @MrGoPoint ? I'd like to investigate your issue and see if I can get a fix in this next release or help you figure out what's going on!

KingOfBrian, 2016-11-22:
Hey @MrGoPoint -- just another ping to see if you need any help

fabiorodella, 2016-12-15:
@KingOfBrian I might be seeing this same issue. After bluetooth is turned off then on again in the device, I can no longer connect to the peripheral, even by trying to call `enableNotify` again. It does get discovered in a scan, but attempting to discover its services also does nothing. Here are the logs:

```
13:36:43.057 🔵DEBUG BluetoothController.init():65 - centralManagerDidUpdateState: - <CBCentralManager: 0x175a7ab00>
13:36:43.058 🔵DEBUG BluetoothController.init():65 - State=5
13:36:57.617 🔵DEBUG BluetoothController.init():65 - scanForPeripheralsWithServices:@[181D] options:(null)
13:36:57.647 🔵DEBUG BluetoothController.init():65 - centralManager:didDiscoverPeripheral:advertisementData:RSSI: - <CBCentralManager: 0x175a7ab00> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
13:36:57.648 🔵DEBUG BluetoothController.init():65 - advertisementData={
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataServiceUUIDs =     (
        181D
    );
}
13:36:57.648 🔵DEBUG BluetoothController.init():65 - RSSI=-44
13:36:57.649 🔵DEBUG BluetoothController.init():65 - (null) discoverServices:@[181D]
13:36:57.651 🔵DEBUG BluetoothController.init():65 - connectPeripheral:89CF787A-F177-45F6-BE49-0F3DD92D7E20 options:(null)
13:36:58.158 🔵DEBUG BluetoothController.init():65 - centralManager:didConnectPeripheral: - <CBCentralManager: 0x175a7ab00> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
13:36:58.159 🔵DEBUG BluetoothController.init():65 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 discoverServices:@[181D]
13:36:58.403 🔵DEBUG BluetoothController.init():65 - peripheral:didDiscoverServices: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 (null)
13:36:58.405 🔵DEBUG BluetoothController.init():65 - Services=@[181D]
13:37:02.339 🔵DEBUG BluetoothController.init():65 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 discoverCharacteristics:@[2A9D] forService:181D
13:37:02.363 🔵DEBUG BluetoothController.init():65 - peripheral:didDiscoverCharacteristicsForService:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 181D (null)
13:37:02.364 🔵DEBUG BluetoothController.init():65 - Characteristics=@[2A9D]
13:37:02.365 🔵DEBUG BluetoothController.init():65 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 setNotifyValue:YES forCharacteristic:2A9D
13:37:02.423 🔵DEBUG BluetoothController.init():65 - peripheral:didUpdateNotificationStateForCharacteristic:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 2A9D (null)
13:37:02.424 🔵DEBUG BluetoothController.init():65 - Notify=YES
13:37:15.125 🔵DEBUG BluetoothController.init():65 - peripheral:didUpdateValueForCharacteristic:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 2A9D (null)
13:37:15.125 🔵DEBUG BluetoothController.init():65 - Value=<205050>
13:37:21.336 🔵DEBUG BluetoothController.init():65 - centralManager:didDisconnectPeripheral:error: - <CBCentralManager: 0x175a7ab00> 89CF787A-F177-45F6-BE49-0F3DD92D7E20 (null)
13:37:21.339 🔵DEBUG BluetoothController.init():65 - connectPeripheral:89CF787A-F177-45F6-BE49-0F3DD92D7E20 options:(null)
13:37:21.879 🔵DEBUG BluetoothController.init():65 - centralManagerDidUpdateState: - <CBCentralManager: 0x175a7ab00>
13:37:21.880 🔵DEBUG BluetoothController.init():65 - State=4
13:37:38.161 🔵DEBUG BluetoothController.init():65 - centralManagerDidUpdateState: - <CBCentralManager: 0x175a7ab00>
13:37:38.162 🔵DEBUG BluetoothController.init():65 - State=5
13:37:53.338 🔵DEBUG BluetoothController.init():65 - scanForPeripheralsWithServices:@[181D] options:(null)
13:37:53.609 🔵DEBUG BluetoothController.init():65 - centralManager:didDiscoverPeripheral:advertisementData:RSSI: - <CBCentralManager: 0x175a7ab00> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
13:37:53.611 🔵DEBUG BluetoothController.init():65 - advertisementData={
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataServiceUUIDs =     (
        181D
    );
}
13:37:53.612 🔵DEBUG BluetoothController.init():65 - RSSI=-46
13:37:53.614 🔵DEBUG BluetoothController.init():65 - (null) discoverServices:@[181D]
13:38:29.499 🔵DEBUG BluetoothController.init():65 - scanForPeripheralsWithServices:@[181D] options:(null)
13:38:29.563 🔵DEBUG BluetoothController.init():65 - centralManager:didDiscoverPeripheral:advertisementData:RSSI: - <CBCentralManager: 0x175a7ab00> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
13:38:29.564 🔵DEBUG BluetoothController.init():65 - advertisementData={
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataServiceUUIDs =     (
        181D
    );
}
13:38:29.566 🔵DEBUG BluetoothController.init():65 - RSSI=-47
13:38:29.566 🔵DEBUG BluetoothController.init():65 - (null) discoverServices:@[181D]
```

KingOfBrian, 2016-12-20:
Hey @fabiorodella -- Thanks for the help. Can you checkout #43 and make sure that it fixes your issue? This was a regression from the past, so I added a test to  ensure it's good in the future.

Thanks

Brian

fabiorodella, 2016-12-21:
@KingOfBrian Yup, that did the trick! Thanks a lot for the quick fix!

KingOfBrian, 2016-12-22:
I pushed up 1.1.3 @fabiorodella -- thanks for your help

closed 2016-12-22 by KingOfBrian