Rightpoint/RZBluetooth #40: CocoaPods issue: RZBTestDefines.h file not found when using "RZBluetooth/Test" pod

cpatterson-lilly, 2016-10-12:
New issue with the 10/10 commit:

I'm using RZBSimulatedTestCase in my unit tests, and it appears that CocoaPods does not pull down the new file RZBTestDefines.h when I include the "RZBluetooth/Test" pod in my test target in my Podfile. Thus my test target build fails.

I'm guessing that line 43 of the podspec should change from:
```ruby
    test.public_header_files = "RZBluetoothTests/RZBSimulatedTestCase.h"
```
to:
```ruby
    test.public_header_files = "RZBluetoothTests/RZBSimulatedTestCase.h", 
                               "RZBluetoothTests/RZBTestDefines.h"
```

KingOfBrian, 2016-10-13:
Good catch, I'll update the podspec. I thought using master as an integration would be an issue, sorry about that. Your `Podfile` should be pointing to a version to protect you from some of this though, are you pointing to `master`?

cpatterson-lilly, 2016-10-13:
Yes, I'm referencing the podspec directly from the repo. My test target looks like this:
```
target 'MyProjectTests' do
    # Pull the podspecs for RZBluetooth directly from its github repo.
    pod 'RZBluetooth',      :git => "https://github.com/Raizlabs/RZBluetooth.git"
    pod 'RZBluetooth/Mock', :git => "https://github.com/Raizlabs/RZBluetooth.git"
    pod 'RZBluetooth/Test', :git => "https://github.com/Raizlabs/RZBluetooth.git"
end
```
But we commit the Podfile.lock and the Pods folder, so we have control over when we take a new release. I'm just working in a Swift 3 branch at the moment.

KingOfBrian, 2016-10-17:
Ok, that makes sense. I'll use an integration branch next time, but I'm glad you caught the error.

KingOfBrian, 2016-11-22:
Hey @cpatterson-lilly -- I just pushed up a new 1.1.1, and it should fix this issue. I'm going to close this, re open if you are still encountering a problem!

closed 2016-11-22 by KingOfBrian