Rightpoint/RZBluetooth #42: Should `enableNotify` continue to work after a peripheral reconnects?

fabiorodella, 2016-12-15:
### Issue
Here's my scenario:

- have a peripheral UUID that was discovered previously
- as soon as the app is launched, obtain a new `RZBPeripheral` instance from `RZBCentralManager` and enable `maintainConnection`
- call `enableNotify` and observe that updated values are received correctly
- power cycle the peripheral
- updated values are no longer received

I can work around this by setting a `connectionDelegate` and having it call `enableNotify` again once the connection succeeds, but should this be needed at all? From the docs I was expecting that the framework would handle this.

### Bluetooth Logs
```
10:40:28.438 🔵DEBUG BluetoothController.init():55 - centralManagerDidUpdateState: - <CBCentralManager: 0x1720616c0>
10:40:28.439 🔵DEBUG BluetoothController.init():55 - State=5
10:40:28.439 🔵DEBUG BluetoothController.init():55 - connectPeripheral:89CF787A-F177-45F6-BE49-0F3DD92D7E20 options:(null)
10:40:28.757 🔵DEBUG BluetoothController.init():55 - centralManager:didConnectPeripheral: - <CBCentralManager: 0x1720616c0> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
10:40:28.758 🔵DEBUG BluetoothController.init():55 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 discoverServices:@[181D]
10:40:28.831 🔵DEBUG BluetoothController.init():55 - peripheral:didDiscoverServices: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 (null)
10:40:28.831 🔵DEBUG BluetoothController.init():55 - Services=@[181D]
10:40:28.832 🔵DEBUG BluetoothController.init():55 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 discoverCharacteristics:@[2A9D] forService:181D
10:40:28.832 🔵DEBUG BluetoothController.init():55 - peripheral:didDiscoverCharacteristicsForService:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 181D (null)
10:40:28.832 🔵DEBUG BluetoothController.init():55 - Characteristics=@[2A9D]
10:40:28.833 🔵DEBUG BluetoothController.init():55 - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 setNotifyValue:YES forCharacteristic:2A9D
10:40:28.861 🔵DEBUG BluetoothController.init():55 - peripheral:didUpdateNotificationStateForCharacteristic:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 2A9D (null)
10:40:28.862 🔵DEBUG BluetoothController.init():55 - Notify=YES
10:40:40.758 🔵DEBUG BluetoothController.init():55 - peripheral:didUpdateValueForCharacteristic:error: - 89CF787A-F177-45F6-BE49-0F3DD92D7E20 2A9D (null)
10:40:40.759 🔵DEBUG BluetoothController.init():55 - Value=<205050>
10:40:45.509 🔵DEBUG BluetoothController.init():55 - centralManager:didDisconnectPeripheral:error: - <CBCentralManager: 0x1720616c0> 89CF787A-F177-45F6-BE49-0F3DD92D7E20 Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.}
10:40:45.512 🔵DEBUG BluetoothController.init():55 - connectPeripheral:89CF787A-F177-45F6-BE49-0F3DD92D7E20 options:(null)
10:40:51.601 🔵DEBUG BluetoothController.init():55 - centralManager:didConnectPeripheral: - <CBCentralManager: 0x1720616c0> 89CF787A-F177-45F6-BE49-0F3DD92D7E20
```

KingOfBrian, 2016-12-15:
Hey @fabiorodella -- your work around is the correct approach. Sorry this was confusing, is there anything in the documentation you can see that confused you here? It's explained in the `Availability Interaction` section -- https://github.com/Raizlabs/RZBluetooth#availability-interactions

I can't really be smarter about this, since on connection behavior can often be complex.

fabiorodella, 2016-12-15:
My bad, there's nothing confusing about the documentation, I guess I confused myself... And right after creating the issue I saw this bit in the documentation for the `enableNotify` method:

```When the connection is torn down, all of the change blocks will be invalidated, and the notify block should be registered when the connection occurs.```

Sorry about this!

KingOfBrian, 2016-12-15:
No problem, glad to help

closed 2016-12-15 by KingOfBrian