Rightpoint/RZBluetooth #45: If a peripheral disconnects while the scan that found it is running, the scan will fail

fabiorodella, 2017-02-07:
### Issue
I've noticed that if I connect to a peripheral that was found in a scan operation, and if the peripheral disconnects from us while the scan is still running, the scan will fail with the disconnection error being passed to the scan block. There are ways to work around this in my code, but this behavior seems weird to me.

### Bluetooth Logs
The disconnection error is logged as follows:
```
centralManager:didDisconnectPeripheral:error: - <CBCentralManager: 0x174265200> 0D789391-1F42-41AB-82D8-3A431C753018 Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.}
```

KingOfBrian, 2017-02-07:
Hey @fabiorodella -- Good catch! I made a PR #46 which should resolve this and add a unit test to confirm things. If you can test it out I'll make a new release later today.

Thanks!

fabiorodella, 2017-02-07:
@KingOfBrian works like a charm!

closed 2017-02-07 by KingOfBrian

KingOfBrian, 2017-02-07:
I just pushed up version `1.1.4` @fabiorodella -- can you do an update and make sure everything looks good?

fabiorodella, 2017-02-08:
@KingOfBrian yup, I updated the pod to `1.1.4`, everything works fine! Thanks for the fix!

PR #46 (2017-02-07): Fix callback issue
This issue fixes #45, where a connection error would complete the scan command.