Rightpoint/RZBluetooth #58: Disconnection, Reconnection and state restoration

mph070770, 2017-04-07:
As mentioned in #57, i'm trying out your framework with the heartrate example. I'm quite new to iOS so it's difficult for me to understand everything.

What I have found is that once I am connected to a device, if I move back from the service list view, the sensor stays connected. I would have expected it to disconnect at that point. When I go back to the "scan" view the device is not visible as it's still connected and therefore not advertising. I don't know if what I am seeing is a consequence of the fixes i put in place for #57? Could you advise please?

What I would like to evaluate, is making my app robust to the device disappearing (temporarily out of range) or my app being closed by iOS due to memory pressure etc. I essentially want my app to be always running unless I intentionally kill it. Can you advise on what I need to do to achieve that with your framework? I think some of the key elements are already there?

Thanks

cpatterson-lilly, 2017-04-07:
Hi @mph070770,

I'm not the owner of the framework, just another user and contributor, but I'll try to help out.

I think all you need to do to achieve what you want when using RZBluetooth is the following:

1. Make sure you have enabled the `"bluetooth-central"` background mode for your app, by either:
    - checking the "Uses Bluetooth LE accessories" checkbox under **Background Modes** in the "Capabilities" tab of your build target's settings, OR
    - Adding the "UIBackgroundModes" key to your Info.plist file whose value is an array containing one string value "bluetooth-central". (Both options essentially do this.)

2. Set the `maintainConnection` flag on the `RZBPeripheral` instance representing your device.

3. Register an observer of the `RZBCentralManagerRestorePeripheralNotification` at app launch so you can automatically get a restored `RZBPeripheral` instance for your device when available.

mph070770, 2017-04-07:
Hi @cpatterson-lilly, thanks for your help. I've got the background mode set aleady, and have added:

`peripheral.maintainConnection = TRUE;`

to initWithPeripheral in PeripiheralViewController.m.  Can you advise how I should register an observer or set a restorationHandler?

Thanks

mph070770, 2017-04-13:
Can anyone help please?

KingOfBrian, 2017-04-13:
Hey @mph070770 , sorry for the delay. It sounds like this should work, BLE restoration should be able to wake up the application for notifications or successful connection attempts. RZBluetooth doesn't change any backgrounding behavior, but it should manage the state restoration for you.

Debugging failed backgrounding is pretty much the worst. One trick that helps is to open the iPhones console (Xcode -> Devices -> Click the down-triangle-in-a-box) and see if the device reports anything. This debug logging varies between iOS versions so I'm not positive it will help you, but it usually logs interesting events. The other approach is to log on the peripheral side. This is easier if you're working with a development board, and may not be possible if you aren't. Between these two, you should be able to figure out what's going on.

My gut says that for some reason the OOM killer is causing the app to disconnect from the peripheral, or there's a bug in the RZBluetooth restoration code. An log in the app delegate should rule out a bug in the restoration code, and looking for other logging messages can probably help figure out how the connection is being managed.

Also, make sure you read through https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW1 -- it may contain something helpful.

KingOfBrian, 2017-04-26:
Hey @mph070770 -- I'd be interested in what you've found. These corner cases can be tricky to diagnose, and document, and I'd love to document whatever you've found, or fix any issues you've encountered.

I'm going to close this issue and assume you've figured out your issue, but if you want to update any information here, I'll work to improve the documentation (or fix any bugs you may of encountered)

closed 2017-04-26 by KingOfBrian