Rightpoint/RZBluetooth #59: retrieveConnectedPeripheralsWithServices not implemented

matsoft90, 2017-04-18:
There is no way for us to retrieve the currently connected peripherals. So for instance, if I am making two apps that use the same peripheral at on the same device I can use `CBCentralManager`'s `retrieveConnectedPeripheralsWithServices:` method but there is no support for that on `RZBluetooth` yet

KingOfBrian, 2017-04-19:
Hey @matsoft90, good catch. I'm not sure when I'll get a chance to write this, but if you wanted to try your hand at adding support it should be pretty straight forward.

- Add a new Command subclass (Probably start with `RZBConnectCommand`).
- Add the method to instantiate the command and configure callback blocks.
- Add the delegate to `RZBCentralManager` and complete the command.

The testing can be done just expanding `RZBMockCentralManager`, but the simulated connection might get a bit tricky. If you want to do the above and make a PR without unit tests I'll merge it and back fill the tests.

Brian

matsoft90, 2017-04-21:
Thanks Brian for the pointers. I'm actually thinking of switching to `RZBluetooth` for a big project and do need `retrieveConnectedPeripheralsWithServices:` as we have multiple extensions that communicate with the peripheral as well so I need it. I haven't looked through your code very extensively yet so I'm not sure if I'm equipped with making this change yet. But I'll look through it and create a PR when I'm done.

cpatterson-lilly, 2017-04-21:
@KingOfBrian Are you sure this API needs to be asynchronous, with an `RZBCommand` subclass? The `CBCentralManager` API is synchronous, returning the array of `CBPeripherals`, without a completion block. It seems to me all that needs to be done is to wrap a simple method around the CB call and convert the result to an `RZBPeripheral` array.

KingOfBrian, 2017-04-24:
Oh, thanks @cpatterson-lilly!  That should super easy then. You can add an extension to RZBluetooth to get the functionality pretty easily, just call the method on `coreCentralManager` and map them to RZBPeripherals.

I'd love a PR with the method and a test though. It should be better encapsulated.

cpatterson-lilly, 2017-04-24:
PR coming soon!

KingOfBrian, 2017-04-26:
Hey @matsoft90 , @cpatterson-lilly has implemented this, and it's available on master. I'm going to make a new release in the next few days, but if you want to try it out, feel free.

I'll close this issue when the new RZBluetooth release is done.

matsoft90, 2017-04-26:
Thank you so much @KingOfBrian @cpatterson

vicktor, 2017-05-24:
Hi  @KingOfBrian @cpatterson-lilly 

is retrieveConnectedPeripheralsWithServices implemented? any ETA for this?

Thanks

KingOfBrian, 2017-05-30:
Hey @vicktor , Check out the latest release: `1.2.1`. Thanks @cpatterson-lilly !

closed 2017-05-30 by KingOfBrian