Rightpoint/RZBluetooth #61: RZBEnableMock not working from Swift Code

chintan1891, 2017-04-24:
I am invoking RZBEnableMock(YES) from "load" method in Objective C. This category is included in a Swift project, but "RZBEnableMock" is not working. It is not able to Swizzle alloc method of CBCentralManager.

cpatterson-lilly, 2017-04-24:
Hi @chintan1891 

I am not the author, just a contributor. We have a Swift project with about 700 unit tests that rely heavily on `RZBEnableMock` and the RZBMockBluetooth code. In our experience, it works fine from Swift.

However, we are simply adding the call to `RZBEnableMock(true)` in our `XCTestCase` class `setUp()` method, like this:

```
    override class func setUp() {
        // Enable mocking of BLE connections by RZBluetooth.
        RZBEnableMock(true)
        
        // Shorten RZBluetooth default timeout value for faster testing
        RZBUserInteraction.setTimeout(0.5)
        
        RZBSetLogHandler { ... }
        
        super.setUp()
    }
```

We also use `RZBSimulatedTestCase` for some of our unit tests, also using Swift.

KingOfBrian, 2017-04-26:
Hey @chintan1891, I'm going to close this issue since we haven't heard back from you. Feel free to reopen this if you need help getting things working, but adding more code that describes how you're using `RZBEnableMock` will help diagnose the issue.

To take a total shot in the dark, I always see issues using `load` with test targets because the timing can be confusing. [This](https://github.com/AliSoftware/OHHTTPStubs/wiki/A-tricky-case-with-Application-Tests) blog post has some good information on that.

closed 2017-04-26 by KingOfBrian