Rightpoint/RZBluetooth #63: sorry about question

vicktor, 2017-05-08:
there is no issue, but, do you have an example of restorationHandler use?

sorry, thanks in advance

KingOfBrian, 2017-05-08:
Hey @vicktor, this should behave exactly like the CBCentralManager callback. If your app is restored by the system, the callback will be passed an array of RZBPeripherals, as described in the documentation:

```An array (an instance of NSArray) of CBPeripheral objects that contains all of the peripherals that were connected to the central manager (or had a connection pending) at the time the app was terminated by the system.
``` 
The CBPeripheral will be wrapped with RZBPeripherals and then the callback block is triggered. Often the next step is to check to make sure that the correct notify blocks are configured, but what needs to be done here varies on the application.

vicktor, 2017-05-08:
Thanks Brian, please can you write some sample code? even if callback is empty.

thanks!

KingOfBrian, 2017-05-08:
Something like this?

```swift
centralManager.restorationHandler = { peripherals in 
  for peripheral in peripherals {
    peripheral.doMyAppSyncThing()
  }
}
```

vicktor, 2017-05-09:
yes!!!!

very thankful!!!

and again, sorry...

KingOfBrian, 2017-05-09:
Good luck!

closed 2017-05-09 by KingOfBrian