Rightpoint/RZBluetooth #64: RZBSimulatedDevice cannot handle same characteristic UUID on multiple services

cpatterson-lilly, 2017-05-17:
### Issue
The `RZBSimulatedDevice` class cannot handle the case where a device implements 2 or more services that share a common characteristic UUID. If you try to add a read/write/subscribe handler for the duplicate characteristic UUID, obviously, it overwrites the previous one added.

The internal `readHandlers`, `writeHandlers`, and `subscribeHandlers` dictionaries use only the characteristic UUID as the key; they should use a combination of characteristic UUID and service UUID (or an `RZBUUIDPath`).

Similarly, the corresponding `{add|remove}{Read|Write|Subscribe}CallbackForCharacteristicUUID:handler:` methods should be modified to take an additional service UUID parameter (or replace them with an `RZBUUIDPath` parameter).

cpatterson-lilly, 2017-05-17:
Just sent a PR to fix this one.

cpatterson-lilly, 2017-05-18:
Just discovered that `RZBPeripheral` has a similar problem with its `notifyBlocksByUUID` dictionary. This also needs to differentiate characteristics with the same UUID on different services.

KingOfBrian, 2017-05-19:
Ouch, good catch. I'll move any remaining feedback to the PR, but at a high level:
- I'd like to keep RZBUUIDPath private
- I'd like to maintain compatibility, so I'd prefer to have the `{add|remove}{Read|Write|Subscribe}CallbackForCharacteristicUUID:handler:` methods still function.

closed 2017-05-19 by KingOfBrian