Rightpoint/RZBluetooth #68: Project deployment target doesn't match Podspec

gilesvangruisen, 2017-06-20:
### Issue

The podspec lists the iOS deployment target at 9.0, but the Xcode project lists the iOS deployment target at 8.4.

Is there any reason for this? Can the podspec's deployment target be dropped down to 8.4, or can they potentially be both dropped even further to 8.0?

Thanks!

KingOfBrian, 2017-06-20:
Oops, that was my mistake! I only aim to keep support for the last two major versions of iOS, but that's just due to our target preference. I don't think there's anything that should prevent 8.0+ support.

Any chance you can try it out and pull together a PR? Should be pretty trivial. If it's not a trivial change, I should be able to look at it in a few days.

Thanks!

gilesvangruisen, 2017-06-20:
@KingOfBrian Sure thing! Just wanted to ask first in case there was any reason behind it.

PR forthcoming :D

cpatterson-lilly, 2017-06-20:
Just FYI, although we just added support for the method `CBCentralManager -retrieveConnectedPeripheralsWithServices:` in v1.2.1, it's OK, because that is supported back to iOS 7.0.

KingOfBrian, 2017-06-20:
Just pushed up `1.2.2` 🎉  -- thanks @gilesvangruisen !

closed 2017-06-20 by KingOfBrian