Rightpoint/RZBluetooth #71: When you call rzb_fetchDeviceInformationKeys it does not actually read the first time

matsoft90, 2017-06-22:
After connection at some point I am calling `rzb_fetchDeviceInformationKeys` to read device information. However, the first call to this always returns nil values as it doesn't actually read the values 

I see the following logs
```
RZBLogLevelCommand - <DEVICE UUID> discoverServices:@[Device Information]

RZBLogLevelDelegate - peripheral:didDiscoverServices: - <DEVICE UUID> (null)

RZBLogLevelDelegateValue - Services=@[180A]

RZBLogLevelCommand - <DEVICE UUID> discoverCharacteristics:@[System ID, Model Number String, Serial Number String, Software Revision String, Firmware Revision String, Hardware Revision String, Manufacturer Name String, PnP ID] forService:180A

RZBLogLevelDelegate - peripheral:didDiscoverCharacteristicsForService:error: - <DEVICE UUID> (null)

RZBLogLevelDelegateValue - Characteristics=@[2A29, 2A24, 2A25]
```
However, if I call it again once the characteristics are discovered it returns the values and I see logs like

```
RZBLogLevelCommand - <DEVICE UUID> readValueForCharacteristic:Model Number String

RZBLogLevelCommand - <DEVICE UUID> readValueForCharacteristic:Serial Number String

RZBLogLevelCommand - <DEVICE UUID> readValueForCharacteristic:Manufacturer Name String

RZBLogLevelCommand - <DEVICE UUID> discoverCharacteristics:@[System ID, Software Revision String, Firmware Revision String, Hardware Revision String, PnP ID] forService:180A

RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE UUID> (null)

RZBLogLevelDelegateValue - Value=<some value>

RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE UUID> (null)

RZBLogLevelDelegateValue - Value=<some value>

RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE UUID> (null)

RZBLogLevelDelegateValue - Value=<some value>

RZBLogLevelDelegate - peripheral:didDiscoverCharacteristicsForService:error: - FF9D814B-B379-46EB-B5A8-F995DFC33A2C 180A (null)

RZBLogLevelDelegateValue - Characteristics=@[2A29, 2A24, 2A25]

```

My understanding was that `RZBluetooth` handles the discovery of services and characteristics internally and I do not need to call `discoverService` and `discoverCharacteristics` methods...

cpatterson-lilly, 2017-06-26:
I see this behavior in our code as well; I just worked around it. I haven't dug in to investigate why it is happening yet.

KingOfBrian, 2017-06-27:
Interesting -- that's not good! Can you include a full log dump?

matsoft90, 2017-06-28:
@KingOfBrian Like I posted above these are the logs I see

When I read it the first time I get
```
RZBLogLevelCommand - <DEVICE_UUID> discoverServices:@[Device Information]
RZBLogLevelDelegate - peripheral:didDiscoverServices: - <DEVICE_UUID> (null)
RZBLogLevelDelegateValue - Services=@[180A]
RZBLogLevelCommand - <DEVICE_UUID> discoverCharacteristics:@[System ID, Model Number String, Serial Number String, Software Revision String, Firmware Revision String, Hardware Revision String, Manufacturer Name String, PnP ID] forService:180A
RZBLogLevelDelegate - peripheral:didDiscoverCharacteristicsForService:error: - <DEVICE_UUID> 180A (null)
RZBLogLevelDelegateValue - Characteristics=@[2A29, 2A24, 2A25]
```

And when I read it the second time

```
RZBLogLevelCommand - <DEVICE_UUID> readValueForCharacteristic:Model Number String
RZBLogLevelCommand - <DEVICE_UUID> readValueForCharacteristic:Serial Number String
RZBLogLevelCommand - <DEVICE_UUID> readValueForCharacteristic:Manufacturer Name String
RZBLogLevelCommand - <DEVICE_UUID> discoverCharacteristics:@[System ID, Software Revision String, Firmware Revision String, Hardware Revision String, PnP ID] forService:180A
RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE_UUID> (null)
RZBLogLevelDelegateValue - Value=<ACTUAL DATA VALUE>
RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE_UUID>(null)
RZBLogLevelDelegateValue - Value=<ACTUAL DATA VALUE>
RZBLogLevelDelegate - peripheral:didUpdateValueForCharacteristic:error: - <DEVICE_UUID> (null)
RZBLogLevelDelegateValue - Value=<ACTUAL DATA VALUE>
RZBLogLevelDelegate - peripheral:didDiscoverCharacteristicsForService:error: - <DEVICE_UUID> (null)
RZBLogLevelDelegateValue - Characteristics=@[2A29, 2A24, 2A25]
```

And for obvious reasons I am obscuring the device UUID and the values being read... As you can see when I call it the first time I do not get any `didUpdateValueForCharacteristic:error:` get called. Only after the first discovery is complete does it get called. This happens consistently and every time. Not only this, if I manually call `readCharacteristicUUID: serviceUUID:completion:` on any of these characteristics, I see the same behavior the first time.

matsoft90, 2017-07-13:
@KingOfBrian It seems it only happens if you call it for characteristics that are not present in the device. So if the device has all of the characteristics, the `peripheral:didUpdateValueForCharacteristic:error:` is called and everything works as expected.

cpatterson-lilly, 2017-07-18:
That matches up with what we saw as well. Our workaround was to only request the device info keys we knew were supported by our device(s).

KingOfBrian, 2017-07-18:
I'm not entirely sure why this would result in all nil given the code, but it just seems like a bug. It'd be great if one of you could put up a PR. I don't have a device with a DeviceInfo Service at the moment, so it'd be a bit of pain to do one. I think explicitly discovering services and then reading all of the discovered services might work better.

matsoft90, 2017-07-20:
@cpatterson-lilly That's exactly what I did. I only requested the ones that are there in my peripherals and it works fine. @KingOfBrian but services discovery is already there so it shouldn't be needed. I'll see if discovering services first changes anything.