Rightpoint/RZBluetooth #73: Add better swift support by using `class` properties, `nullable`, `nonnull` and `NSDicationary` and `NSArray` with types

matsoft90, 2017-07-13:
Right now `RZBluetooth` is great for use in Objective-C but when you use it in Swift you see a lot of `[Any]!`,  `[AnyHashable: Any]` and `String!` and also `class func` instead of `var`.

To improve this what needs to be done is

- Add `nullable` and `nonnull` to properties and method params where applicable. `(nonatomic, nullable, copy)`
- Add type to `NSArray` and `NSDictionary` properties, e.g. `NSArray <NSString *>* 
- Replace many of the class methods like `+ (CBUUID *)serviceUUID;` to `@property (nonatomic, strong, readonly, nonnull, class) CBUUID *serviceUUID;` 

Doing all this will make the library look and feel native to Swift when used. I know it is a massive change and will require a major version bump and migration notes. 

@KingOfBrian do you think this is a good change and should be done?

KingOfBrian, 2017-07-18:
This sounds like a great change @matsoft90 ! We did an initial pass for swift interop, and caught a number of them, but if there are any we missed, I'd love a PR to fix it.

One thing to note, is it would be great to do all of them in one pass. It changes the public interface pretty drastically for Swift and would require a major version bump.

Also, just got back from vacation, sorry for the delay!

matsoft90, 2017-07-20:
@KingOfBrian I'm currently actively engaged in a project and as it is a major thing to find them all and then a breaking change as a whole I'll do this when I get some free time.

KingOfBrian, 2018-05-15:
I believe we're in good shape nullability wise, so I'm going to close this out. If you have any issues with nullability, please log specific issues.

closed 2018-05-15 by KingOfBrian