Rightpoint/RZBluetooth #74: A peripheral always reconnects after cancelConnectionWithCompletion: call even when maintainConnection is false

matsoft90, 2017-07-18:
@cpatterson-lilly @KingOfBrian I am having a small problem... I have a peripheral that always automatically reconnects if I manually call `cancelConnectionWithCompletion:`. I have `maintainConnection` false for this peripheral right now. Since `maintainConnection` is false the `connectWithCompletion:` method is not called from within `connectionEvent:error:`

I see the following in the logs

```
RZBLogLevelCommand - cancelPeripheralConnection: <PERIPHERAL UUID>
RZBLogLevelDelegate - centralManager:didDisconnectPeripheral:error: - <CBCentralManager: 0x17007dd80> <PERIPHERAL UUID> (null)
RZBLogLevelCommand - connectPeripheral:<PERIPHERAL UUID> options:(null)
RZBLogLevelDelegate - centralManager:didConnectPeripheral: - <CBCentralManager: 0x17007dd80> <PERIPHERAL UUID>
```

matsoft90, 2017-07-18:
It appears to be happening inside `connectedPeripheralForUUID:triggeredByCommand` because of a `RZBNotifyCharacteristicCommand` I do have notify enabled for a couple characteristics. But even then a disconnection should cancel that as well.

matsoft90, 2017-07-18:
I was calling `clearNotifyBlock` on my disconnected peripheral. I'm sorry to bother you both.

KingOfBrian, 2017-07-18:
No problem @matsoft90. I'm still interested in the usability here and this still feels a bit off. You called `clearNotifyBlock` on a disconnected peripheral, and it caused a reconnection? If so, that's not entirely intuitive, and a valid bug in my opinion!

matsoft90, 2017-07-20:
Yes @KingOfBrian that's exactly what I did. And you're right. It isn't very intuitive. A solution could be to not call the reconnect method if the command is of type `RZBNotifyCharacteristicCommand`. I'll see if I can make this quick change and create a PR.

matsoft90, 2017-07-24:
@KingOfBrian I created a PR though I'm not sure if this is how you wanted it implemented.

KingOfBrian, 2018-11-09:
This should be addressed in #95 !

closed 2018-11-09 by KingOfBrian