Rightpoint/RZBluetooth #80: RZBEnableMock()  doesn't work in Swift

cpatterson-lilly, 2018-01-12:
It seems that `RZBEnableMock()` doesn't work when used in pure Swift code.

This came up while we were writing some peripheral simulator unit tests.

Try pasting the following code into the file `SwiftTestCase.swift`:

```swift
class SwiftTestCase2: XCTestCase {
    
    func testMocking() {
        var centralManager = CBCentralManager()
        XCTAssertNil(centralManager.mock)
        
        var peripheralManager = CBPeripheralManager()
        XCTAssertNil(peripheralManager.mock)

        // Enable RZBluetooth mocking
        RZBEnableMock(true)
        
        centralManager = CBCentralManager()
        XCTAssertNotNil(centralManager.mock) // Fails
        
        peripheralManager = CBPeripheralManager()
        XCTAssertNotNil(peripheralManager.mock) // Fails
        
        // Restore normal operation
        RZBEnableMock(false)
        
        centralManager = CBCentralManager()
        XCTAssertNil(centralManager.mock)
        
        peripheralManager = CBPeripheralManager()
        XCTAssertNil(peripheralManager.mock)
    }
}
```

Apparently, Swift does not call the Obj-C `alloc` method?

mplorentz, 2018-01-19:
I have been able to work around this issue by creating an Objective-C factory class that has functions to instantiate the Core Bluetooth classes that I need. I call this factory class from my Swift code. It's no pretty but it works.

KingOfBrian, 2018-05-15:
Yea, there's not much we can do here. It might make sense to add some factory methods to RZBluetooth so the fix is easy at the least. Any thoughts?

cpatterson-lilly, 2018-05-15:
How about replacing `RZBEnableMock()` with an `RZBMocking` class that wraps closures with mocked objects. Something like `Data.withUnsafeBytes()`, etc.

Something like this (in Swift)?

```swift
class RZBMocking {
    public func withMockCentralManager(_ block: (RZBMockCentralManager)->()) {
    }
    public func withMockPeripheralManager(_ block: (RZBMockPeripheralManager)->()) {
    }
}
```

KingOfBrian, 2018-11-09:
I'm going to add a `make()` method to work around this. I'm defining it in `RZBMockFramework` -- would that of made it available in your use cases, or does it want to be in RZBluetooth?

cpatterson-lilly, 2018-11-09:
I assume you mean a class method called `make()` on the `RZBMockPeripheralManager` and `RZBMockCentralManager` classes, which live in the RZBMockFramework?
I see no problems with that solution.

KingOfBrian, 2018-11-09:
It's actually implemented in CBCentralManager so it isolates knowledge of the mocks. See #95.

closed 2018-11-09 by KingOfBrian