Rightpoint/RZBluetooth #86: Regression calling stopScan in RZBSimulatedTestCase context between 1.0.2 and 1.2.2

drdaz, 2018-05-15:
### Issue

Hi there

I recently upgraded a project I'm working on from 1.0.2 to 1.2.2, and tried to run my test suite on my BT logic. One of my tests (which simulates a connection timeout) crashes in RZBSimulatedCallback.m at line 118.

Here's my test to reproduce. in an RZBSimulatedTestCase:

```
-(void)testRZBFail
{
    [self.device addHopGateService];
    
    [self.centralManager scanForPeripheralsWithServices:@[[CBUUID hopspotsServiceUUID]] options:nil onDiscoveredPeripheral:^(RZBScanInfo * _Nullable scanInfo, NSError * _Nullable error) {
        
    }];
    
    XCTestExpectation *expect = [self expectationWithDescription:@"Complete"];
    
    [[BFTask taskWithDelay:5] continueWithBlock:^id _Nullable(BFTask * _Nonnull t) {
        [self.centralManager stopScan];
        [expect fulfill];
        return nil;
    }];
    [self waitForExpectations:@[expect] timeout:10];
    
}
```

If you're not familiar with Bolts / BFTask, it's an implementation of Futures / Promises. What I'm doing here is waiting 5 seconds after we start scanning to call stopScan; if I call it instantly we don't crash.

The addHopGateService method is in a category on RZBSimulatedDevice. It looks like this:

```
-(void)addHopGateService
{
    CBMutableService *hopGateService = [[CBMutableService alloc] initWithType:[CBUUID UUIDWithString:hopspotsServiceUUIDString] primary:YES];

    CBMutableCharacteristic *hopSpotEventNotifyCharacteristic = [[CBMutableCharacteristic alloc] initWithType:[CBUUID UUIDWithString:hopspotsReceiveCharacteristicUUIDString]
                                                                                                   properties:CBCharacteristicPropertyRead | CBCharacteristicPropertyIndicate | CBCharacteristicPropertyNotify
                                                                                                        value:nil
                                                                                                  permissions:CBAttributePermissionsReadable];
    CBMutableCharacteristic *hopSpotControlCharacteristic = [[CBMutableCharacteristic alloc] initWithType:[CBUUID UUIDWithString:hopspotsTransmitCharacteristicUUIDString]
                                                                                               properties:CBCharacteristicPropertyWrite
                                                                                                    value:nil
                                                                                              permissions:CBAttributePermissionsWriteable];
    hopGateService.characteristics = @[hopSpotEventNotifyCharacteristic, hopSpotControlCharacteristic];
    [self addService:hopGateService];
}
```

This should be enough to reproduce the error. Let me know if I can help.

KingOfBrian, 2018-05-15:
Can you add a backtrace of the crash and the output of `RZBSetLogHandler`?

KingOfBrian, 2018-05-15:
Actually, I think this is a dupe of #84 , can you confirm?

KingOfBrian, 2018-05-15:
Something weird is going on, it looks like it's inside `cancel`. Can you try this implementation out? It should be the same, but I'm not sure why releasing the timer is blowing up now. This change will probably blow up, but it might blow up differently.

```
- (void)cancel
{
    @synchronized(self) {
        NSArray *timers = [self.timers copy];
        [self.timers removeAllObjects];
        for (dispatch_source_t timer in timers) {
            dispatch_cancel(timer);
        }
    }
}
```

drdaz, 2018-05-15:
```
- (void)cancel
{
    @synchronized(self) {
        NSArray *timers = [self.timers copy];
        [self.timers removeAllObjects];
        for (dispatch_source_t timer in timers) {
            dispatch_cancel(timer);
        }
    } <---- Now it blows up here. Error looks the same (EXC_BAD_INSTRUCTION 0x0)
}
```

drdaz, 2018-05-15:
In the original implementation the crash looks like this:
[image]

KingOfBrian, 2018-05-15:
Can you try this dispatch method? This is super-paranoid about doing everything inside @synchronized.

```
- (void)dispatch:(RZBSimulatedCallbackBlock)block
{
    NSParameterAssert(block);
    @synchronized(self) {
        NSError *injectError = self.injectError;
        dispatch_source_t timer = dispatch_source_create(DISPATCH_SOURCE_TYPE_TIMER, 0, 0, self.queue);
        uint64_t delay = self.delay * NSEC_PER_SEC;
        dispatch_source_set_timer(timer, dispatch_time(DISPATCH_TIME_NOW, delay), 0, 0);
        __weak typeof(self) weakSelf = self;
        __weak dispatch_source_t weakTimer = timer;

        dispatch_source_set_event_handler(timer, ^{
            [weakSelf triggerBlock:block injectedError:injectError timer:weakTimer];
        });

        [self.timers addObject:timer];
        if (!self.paused) {
            dispatch_resume(timer);
        }
    }
}
```

KingOfBrian, 2018-05-15:
Also, can you run with 'Thread sanitizer' ?
[image]

drdaz, 2018-05-15:
Same result it seems. With both implementations of cancel.

KingOfBrian, 2018-05-15:
Does this always occur in the same exact place @drdaz ? Also, any feedback here @cpatterson-lilly ? I believe this is the same bug you ran into a while ago.

drdaz, 2018-05-15:
It seems to, yes. Do you need more code for a reproducer?

drdaz, 2018-05-15:
I'm building you a reproducer project. Give me a moment :)

KingOfBrian, 2018-05-15:
Don't worry about it @drdaz -- The unit tests reproduce it. It looks like something changed in dispatch_timer. I think I have an approach to do cancellation slightly differently that should avoid the issue.