Rightpoint/RZBluetooth #88: Build warnings using CocoaPods

drdaz, 2018-05-17:
I recently upgraded my xcodeproj gem, which has made building projects using CocoaPods much more... sensitive. I now (and perhaps previously) have the following warnings from RZBluetooth.

Most of it can be fixed by updating the deployment target of RZBluetooth to 7.1 I think.

[image]

drdaz, 2018-05-17:
Alright that was my bad; those warnings were from version 1.0.2.

But the story is similar with the current release. Deploy target needs to be iOS 10 or above now.

drdaz, 2018-05-17:
... and my project needs to support iOS 9. Egads.

drdaz, 2018-05-17:
#89

KingOfBrian, 2018-05-18:
I don't see anything forcing it to be 10.0 and above. What are the warnings for the current release you're seeing? I'd rather not bump to 10+ unless there's a reason.

drdaz, 2018-05-18:
[image]

cpatterson-lilly, 2018-05-18:
Ah, yes, we see this build warning as well. It is an incorrect warning, because RZBluetooth has backward-compatibility with the previous `CBCentralManagerState` and `CBPeripheralManagerState` enums built-in. I assume Xcode just flags any occurrence of the symbol as a warning.

We silenced this warning in our builds using this code in our Podfile:
```ruby
post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == "RZBluetooth.common" || target.name == "RZBluetooth.common-Test" then
        target.build_configurations.each do |config|
            # Silence erroneous RZBluetooth build warnings about `CBManagerState` partial availability
            config.build_settings['CLANG_WARN_UNGUARDED_AVAILABILITY'] = 'NO'
        end
    end
  end
end
```

Perhaps this build setting could be built-in to the RZBluetooth project?

KingOfBrian, 2018-11-09:
Should be addressed in #95

PR #95 (2018-11-09): Pre Release Cleanup
Cleaning up some issues before releasing a new version. Going to bundle into one PR to simplify things, but this resolves #88, #80, #76, #67
closed 2018-11-09 by KingOfBrian