Rightpoint/RZBluetooth #98: Crash in -[RZBPeripheral cancelAllCommands] due to infinite loop

joshuatbrown, 2019-01-21:
### Issue
In the completion block of `-[RZBPeripheral connectWithCompletion:]`, call `-[RZBPeripheral cancelConnectionWithCompletion:]`. It'll create an infinite loop and eventually crash with an EXC_BAD_ACCESS.

For example:

```objc
- (void)testCancelConnectionInConnectCompletion {
    RZBPeripheral *peripheral = [self.centralManager peripheralForUUID:self.connection.identifier];

    [peripheral connectWithCompletion:^(NSError * _Nullable error) {
        [peripheral cancelConnectionWithCompletion:nil];
    }];
    
    [self waitForQueueFlush];
}
```

joshuatbrown, 2019-01-21:
Unfortunately, our fix for this causes a different RZBluetooth test to fail. Here's our fix in `RZBPeripheral`:

```objc
- (void)cancelAllCommands
{
    NSError *error = [NSError errorWithDomain:RZBluetoothErrorDomain
                                         code:RZBluetoothConnectionCancelled
                                     userInfo:@{}];

    NSArray *commands = self.dispatch.commands.copy;
    for (RZBCommand *command in commands) {
        if (command.isExecuted == YES && command.isCompleted == NO) {
            // We are in the completion handler of this command. Do not complete it or we will loop infinitely.
            continue;
        }

        [self.dispatch completeCommand:command
                            withObject:nil error:error];
    }
}
```

This fixes the test case above, but it causes `-[RZBSimulatedTests testConnectionAndCancelWhileNotConnectable]` to fail, and I'm not sure why. Can someone explain the purpose of that test and why it fails with this fix?

cpatterson-lilly, 2019-01-21:
@joshuatbrown It seems that every access to `RZBCommandDispatch.commands` must be `@synchronized`.... Try replacing `self.dispatch.commands.copy` with a call to the (currently internal) method `RZBDispatch.synchronizedCommandsCopy()`, or wrapping that line in an `@synchronized` block?

joshuatbrown, 2019-01-22:
@cpatterson-lilly I replaced my `NSArray *commands = ...` line above with the following:

```objc
NSArray *commands;
@synchronized(self.dispatch.commands) {
    commands = [self.dispatch.commands copy];
}
```

The same test still fails for the same reason. I can leave the `@synchronized` if that's the right way, but it looks like we still need to figure out how to make the test pass.

joshuatbrown, 2019-01-23:
You can see my branch/commit above where I added the unit test and fixed the crash. But I'm still seeing the failure in `-[RZBSimulatedTests testConnectionAndCancelWhileNotConnectable]`. Any idea what's going on?