Rightpoint/RZBluetooth #99: RZBPeripheral.cancelConnection cancels commands for all peripherals

fluehre, 2019-02-13:
### Issue
When you call cancelConnection on a peripheral, it first cancels pending commands. However it does this for all peripherals, not just the one for which we are canceling the connection. In my particular case this was making it impossible to disconnect several devices at a time because they would cancel each other's disconnect commands.

Here's the current code in RZBPeripheral.m:
```obj-c
- (void)cancelAllCommands
{
    NSError *error = [NSError errorWithDomain:RZBluetoothErrorDomain
                                         code:RZBluetoothConnectionCancelled
                                     userInfo:@{}];

    for (RZBCommand *command in [self.dispatch commands]) {
        [self.dispatch completeCommand:command
                            withObject:nil error:error];
    }
}

```

I believe it needs to be changed to the following:
```obj-c
- (void)cancelAllCommands
{
    NSError *error = [NSError errorWithDomain:RZBluetoothErrorDomain
                                         code:RZBluetoothConnectionCancelled
                                     userInfo:@{}];

    RZBUUIDPath *path = RZBUUIDP(self.identifier);
    NSArray *commands = [self.dispatch commandsOfClass:nil matchingUUIDPath:path];

    for (RZBCommand *command in commands) {
        [self.dispatch completeCommand:command
                            withObject:nil error:error];
    }
}

```

KingOfBrian, 2019-02-21:
Great catch! Any chance you can pull together a PR? A test case would be great, but this is pretty problematic. Also, what version are you using?

fluehre, 2019-02-28:
Yeah, I'll try to get a PR with test case together shortly. I'm on 2.0.0-pre.

PR #101 (2019-04-18): Fix cancelAllCommands canceling for all peripherals
Fixes #99. cancelAllCommands is canceling for all peripherals.
Fix to only do so for the current peripheral.
closed 2019-04-18 by chrisballinger